jakobwilm/slstudio #2: Calibration problem

negantia, 2016-05-26:
Hi,

I am going to perform the 3d scanning of the object using the slstudio in educational purposes.
I use the Imaging Source camera DMK22BUC03 (mono),Panasonic LCD Projector PT-L520E, and computer with processor 4x AMD A8-6600K APU with Radeon(tm) HD Graphics and operating system - Ubuntu 15.04 (x86_64). 
Following the provided readme file I have successfully compiled the sources, having previously installed all the dependencies. 
So, I have Qt version 4.8.6, QMake 2.01a, OpenCV 2.4.9, PCL 1.7  and VTK  5.8. 
In order to make it work with my camera I wrote specified camera class containing the methods, which override the virtual ones of Camera.h class and finally can see a good mono picture in calibration window.
I have connected the projector to system unit using the VGA cable, monitor to system unit using a DVI one.
 And try to perform calibration of camera and projector using the provided software and having the following settings: 
Preferences->Pattern Mode - 2x3 Pattern Phase Shift, Camera - Imaging Source DMK22BUC03, Trigger Mode - Software. 
Everything I have in projector checkbox is 1) Screen 0: 1280x1024 , 2) SLStudio Virtual Screen 3) LC3000 API and 4)LC4500 API. 
However if I check the #1 I'll see only the white screen after pressing the "Perform Calibration" button (if I don't use projector, and 2 white screens if I use it and 2 monitors DVI and VGA with help of ARandR). If I check #2 I'll see the stream from my camera but no patterns appear after "snap"
 (as they do in your video https://www.youtube.com/watch?v=swszXuPxGZI ). 

Can you recommend something?

Looking forward to your answer. 
Thank you in advance.

jakobwilm, 2016-09-21:
Hello negantia,
please excuse me for not answering before now. Are you still having issues? Per default, slstudio builds with OpenGLContext.Unix.cpp, which uses old-fashioned XScreens to separate the projection display from your window manager. It requires you to set up the projector as a separate second X screen. If you compile with OpenGLContext.GLFW.cpp instead, the XRandR method is used instead. Please try doing this.
Regards, Jakob

closed 2016-09-21 by jakobwilm

yangjingsen, 2019-04-07:
Hi, I have the same problem with screen,  I only have 1) Screen 0: 1980x1080  2) SLStudio Virtual Screen 3) LC3000 API and 4)LC4500 API, but I have connected with a HDMI projector, what can I do to get "Screen 1"?
Thank for your answer.