jakobwilm/slstudio #5: Does it generate a 3D Mesh?

gree303, 2016-09-20:
Hi Duke, does it generate a 3D Mesh on the fly? What's the output file of this after triggering the different lightproejctions? thx

ereh, 2016-09-21:
Here are some quick answers to your questions:

1. No mesh, nor output files.
2. The output is a real time 3D point cloud.
3. The DLP® LightCrafter™ 4500 (DLPLCR4500EVM) has been used successfully.
4. This PRO6500 from Wintech Digital (https://goo.gl/kqn2IY) might be a good alternative. (Not tested)  I am not sure what you mean by "better output"?

Feel free to submit a pull request f.x if you implement a "point-cloud to mesh" functionality.

gree303, 2016-09-21:
Thanks for the note. Point-cloud is just fine for me.
The LC 4500 has a resolution of 1280x720. In terms of price I was looking for a cheaper option; like this one: [Optoma + 12v trigger](https://www.amazon.com/Optoma-HD26-1080p-Theater-Projector/dp/B00LL2SHK6?psc=1&SubscriptionId=AKIAILSHYYTFIVPWUY6Q&tag=duckduckgo-d-20&linkCode=xm2&camp=2025&creative=165953&creativeASIN=B00LL2SHK6)

Do you think this could work with the 12v trigger? I know it's just a guess, I won't blame you if it won't work...

jakobwilm, 2016-09-21:
The projector would work in software-triggered mode. There is no hardware triggering functionality for pattern projection on consumer projectors. Please read and understand the README. Thank you for not blaming me.

closed 2016-09-21 by jakobwilm