jakobwilm/slstudio #7: Projector question

nikkadim, 2016-10-17:
Hi Jacob,
First of all thank you  for all your hard work, I'm trying to follow your steps, and I have question about Projector setup.
I have:
- TI DLPLCR4500EVM (same as you have) connected via USB (can see it via lsusb)
- Ubuntu 16.04.1 LTS (Release:  16.04)
- Basler USB3 camera (will make pull request as soon as it will start work all together)

but when I run ProjectorTest from ~/slstudio/slstudio-master/src/projector it's shows 
`Screen 0: width 2624, height 900  
terminate called  after throwing an instance of 'char  const*'
also wondering should I upgrade the projector firmware first and may be presetup it? Because right now it shows patterns in red.

jakobwilm, 2016-10-18:
You need to follow the instructions for X Screens setup in the readme.

closed 2016-10-18 by jakobwilm

nikkadim, 2016-10-20:
Sorry, but I still have a problem with Lightcrafter4500 connected via USB/API, I have latest FW (v3.0.0) and I cant see it (have a right udev rules):
```
Bus 002 Device 003: ID 2676:ba02 Basler AG ace
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0451:6401 Texas Instruments, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```
and SLstudio can communicate with projector too, 
`ProjectorLC4500: preparing LightCrafter 4500 for duty...`
but it looks like I missed something and it needs to pre-setup projector somehow!?

[image]
because when I start calibration process all i see is the flickering TI logo in red like that:
[image]

Thank you for your help.

ereh, 2016-10-20:
When using the LC4500 API, the patterns for the Pattern Mode you plan to use need to be stored in the firmware of the projector.  Currently, this is a manual process and the TI e2e forum is the place to get support in that regard.

Instead of choosing "LC4500 API" in the Preferences, you just select the 'Screen' that is presented as the projector. Then run the projector in Pattern Sequence mode (using the Lightcrafter GUI from TI).