jakobwilm/slstudio #8: Strange calibration results

nikkadim, 2016-10-19:
Hi Jakob,
I've tried to calibrate it with a commercial projector and Basler USB camera, calibration went ok, but after that "2+1 Patter Phase shift" option gives me really wavy (instead of flat) surface. Do you have any ideas why it could  happened?
[image]
Thanks.

jakobwilm, 2016-10-19:
You should ensure that you have a linear projector-camera response. With commercial projectors, there usually is gamma setting. You might be able to compensate for that in your graphics driver.

closed 2016-10-19 by jakobwilm

ereh, 2016-10-20:
Same applies to the camera. Check its gamma setting

Also make sure that the camera exposure time is an integer multiple of the projection time.

nikkadim, 2016-10-20:
Thank you, @ereh  @jakobwilm.
I did check projector and camera gammas - it's linear. Did not find anything critical in _nvidia-settings_ too, gamma also linear.

Honestly for me this shape more looks like camera un-synchronized with projector, because the wavy shapes repeat the patterns. @ereh  also mention projection time, how I can get the projection time, for HDMI setup? I thought it's equal to exposure time I set in **Preferences**, am I wrong?

ereh, 2016-10-20:
Most likely your projector is operating at 60 Hz. The projection time will then be t = 1/60 = 16.667 ms.

So in that case, the minimum camera exposure required is 16.667 ms.  That will make sure the entire projected frame is captured. 

If longer exposures are required, then make the exposure time an integer multiple of 16.667 ms. 

33.33 or 50 or 66.666  ....etc.

nikkadim, 2016-10-25:
sorry to bother you @ereh , @jakobwilm 
but unfortunately changing exposure time does not help at all. Shape still the same. Also tried with USB3 Mono8, PointGrey Camera as @jakobwilm  did, same results even when I switch it to RAW8 mode as PointGrey Support Team recommended (to disable all image enhancement).
But what I've noticed with calibration is the camera calibrations is always good error is  ~0.3, but for projector calibration it's ~50-70 (please see screenshot below)

[image]

here is my camera setup:
[image]
projector works at 60Hz via HDMI (from NVidia card) all dynamic features disabled.

below the log with 1 calibration image and single acquisition with 2x3 Pattern Phase Shift:
```
./SLStudio
Point Grey Research  Chameleon3 CM3-U3-31S4M  16306750
        Shutter: 8.32844ms
        Gain: 0dB
GLX-Version 1.4
GL-Version 4.5
OpenGLContext is direct
Framebuffer Complete
Framebuffer Complete
Framebuffer Complete
Framebuffer Complete
Framebuffer Complete
Framebuffer Complete
Framebuffer Complete
Framebuffer Complete
Framebuffer Complete
Framebuffer Complete
Framebuffer Complete
Framebuffer Complete
QObject::killTimer(): Error: timer id 1 is not valid for object 0x1ba63f0 (SLCalibrationDialog, SLCalibrationDialog), timer has not been killed
========================================
Camera Calibration:
- cam_error:
0.3237
- Kc:
[4677.4941, 0, 1023.5;
  0, 4677.4941, 767.5;
  0, 0, 1]
- kc:
[-0.2255, 0, 0, 0, 0]
Projector Calibration:
- proj_error:
55.27
- Kp:
[76576.852, 0, 995.82312;
  0, 76576.852, 459.26981;
  0, 0, 1]
- kp:
[-5860, 0, 0, 0, 0]
Stereo Calibration:
- stereo_error:
38.99
- Rp:
[0.84747833, 0.068889029, -0.5263409;
  -0.53061575, 0.081770569, -0.84365898;
  -0.015079657, 0.99426752, 0.10585239]
- Tp:
[39.39, 57.41, 540.5]
Seq error 1 cam:0.2903 proj:48.72
Releasing OpenGL Context
Point Grey Research  Chameleon3 CM3-U3-31S4M  16306750
GLX-Version 1.4
GL-Version 939515760.0
OpenGLContext is direct
Framebuffer Complete
Framebuffer Complete
Framebuffer Complete
Starting capture!
        Shutter: 8.328ms
        Gain: 0dB
Scan worker 279ms
Releasing OpenGL Context
[0, 1.137e-13]
Decoder: 339ms
Triangulator: 209ms

```
Image sequence with 2x3 Pattern Phase Shift:
[image]

and results:
[image]

I would appreciate any help.

jakobwilm, 2016-10-25:
Hi Nikolay,
the calibration obviously goes wrong. It needs to be fixed before anything else. The "result images" in the calibration dialog are supposed to be "flat illumination images". In your case. There is stripes (blue arrow). You either do not have a linear gray-value response or the images are saturated. Have you run the gamma-check Matlab-script? It needs to show a very straight line.
Regards Jakob

jakobwilm, 2016-10-25:
If you want to use the LC4500 API, the projected patterns need to be uploaded and baked into firmware. Please follow TI’s documentation.
I would suggest that you start using separate X screen configuration OR GLFW and software triggering to begin with.
On most consumer projectors, it is not possible to set gamma to 1.0. You could use the matlab script to check the response.
Please inspect the camera frames to identify whether you have bad synchronization. (View -> Camera Frames).
During debugging, you also might want to write intermediate results to disk. You can use cvtools::writeMat for that.

nikkadim, 2016-10-27:
@jakobwilm  thank you for your reply!
Unfortunately I do not have MATLAB now, do you have ideas how I can do this manually (maybe in ImageJ)?

I just tried do the same with LightCrafter 4500 connected via HDMI with separate X screen, absolutely the same vertical pattern after phase unwrapping and you can't see these patterns in each individual images.
[image]

[image]

[image]

nikkadim, 2016-10-27:
> If you want to use the LC4500 API, the projected patterns need to be uploaded and baked into firmware. Please follow TI’s documentation.

Can I use patterns which LCr4500_3D_Scanner_Application_v2 (TIDA-00254 3D Scanner Softwa) app has? 
Thank you.

jakobwilm, 2016-10-27:
* Your camera images seem over-exposed (saturated). Structured light will never work with saturated images. Please adjust shutter, aperture or light intensity accordingly. The white regions of your checkerboard should have image values ~200.
* If you are using LC4500 in "Video Mode", it applies gamma correction on the projector side. Refer to  TI documentation for the specific value. If you run it in "Pattern Sequence Mode", then no gamma correction, re-interpolation (to diamond pixel array) or "proprietary image enhancement" is performed at the projector.
* Please also check the gamma of your camera.

nikkadim, 2016-11-09:
Thank you @jakobwilm and @ereh! As @jakobwilm  said it was a gamma related issue, by changing NVidia driver gamma value I was able to get almost clean picture and reconstruct flat calibration plate. 
[image]

It feels like I should better use LC4500 in pattern mode where gamma disabled by default, but have no idea ho to do this so far. Do you have may be a tips how what patterns should I load? I've tried procedures in manual, but without good results so far. May be you already have an FW for LC4500 with this patterns you can share?

BTW, to be sure that no  "proprietary image enhancement"  was performed for PointGrey camera they (support team) recommend to get images in **RAW8** mode instead of **MONO8**:

`format7Settings.pixelFormat = FlyCapture2::PIXEL_FORMAT_RAW8;`
instead of 
`format7Settings.pixelFormat = FlyCapture2::PIXEL_FORMAT_MONO8;`

Anyways thank you again for you patience and help, I really appreciate that, and I hope I can make a commit soon with _Basler_ camera support code!