jakobwilm/slstudio #10: LC4500 Pattern Mode with libusb

nikkadim, 2016-11-11:
Hi @jakobwilm, can I ask you for tips about what patterns (sequence) should I load for LC4500 to be able to run your software in pattern mode? I know the procedure to make patterns with DLP Control Software, but I believe your software utilize particular pattern sequence with particular timeouts (exposure and period).
May be you can share the screenshots DLP Control Software with sequence settings or you have a f/w for LC4500 with this patterns you can upload to your this project? 
Thank you in advance.

jakobwilm, 2016-11-11:
Hello Nok,
first of all, we should be very accurate in what we mean by 'pattern sequence mode'. In your case, it appears that you want to store patterns in the firmware. While we have had some success with this approach, it is not the recommended way of operation (as mentioned in the README). In part this is because synchronization (which camera frame represents which pattern?) is difficult if the projector is operating autonomously. The other reason is that a change of patterns requires you to re-flash the firmware on LC4500. A third reason is that this makes it very complicated to correct for projector lens distortion (the patterns need to be pre-distorted before upload. Distortion parameters might vary between calibrations).
What you need to do is to choose a codec, write its patterns to disk (uncomment appropriate lines in the scanworker class), bake these into firmware, modify the LC4500 projector class code and choose exposure and timing values suitable for your specific camera (obeying its frame rate, trigger delay, time for frame transfer).
Regards Jakob

closed 2016-11-11 by jakobwilm

nikkadim, 2017-07-31:
Hello Jakob,

Couple questions if you do not mind.

What PC (CPU/RAM ) did you use with this project for your default Ximea camera? Tried with PointGrey Grasshopper (2048x1536) on i5-6500 CPU @ 3.20GHz (Ubuntu 16.04) and PhaseShiftNStep codec -  looks like one of thread eats all 99%, is it ok?

Second question. The Software mode works really good, thank you!
I'm trying to utilize hardware triggering.
1) Select a codec N Step Phase Shift
2) Dump the all 12 patterns (pat_X.bmp) from SLScanWorker.cpp
[image]
3) Made a new firmware based on this patterns and 3.0.0 firmware
(1) pat_0.bmp + pat_1.bmp  + pat_2.bmp (R G B)
(2) pat_3.bmp + pat_4.bmp  + pat_5.bmp  (R G B)
(3) pat_6.bmp + pat_7.bmp  + pat_8.bmp  (R G B)
(4) pat_9.bmp + pat_10.bmp  + pat_11.bmp (R G B)
[image]
(5) Saved and pload this new firmware
(6) Made a pattern sequence based on the images in flash (in the same order) with Exposure=50ms:
[image]
(7) Matched projector pattern exposure time to camera exposure, connected them with external trigger cable, and switched to Hardware mode. Also just in case enable 2x2 binning on camera to reduce computation. 
As a result I'me getting cloud but instead of water bottle cloud I see distorted cloud:
[image]
Could you please suggest anything? May be I missed some step.