jakobwilm/slstudio #12: Hardware Trigger

nikkadim, 2016-11-16:
I've a sync cable between camera and **LightCrafter 4500** and I switched **SLStudio** to _Hardware Triggering_, but from projector output (_TRIG_OUT1_) I see no pulse at the scope to trigger the camera. In a project README file you also mention that for realtime setup we have to use  _/tools/lc4500startup_ to prepare projector, but I did not find _LCR_SetTrigOutConfig_ function which makes output setup. Am I missed something? Thank you.
[image]

PS: Another question - why grey-white checkerboard matters? Why not black-white?
T

jakobwilm, 2016-11-16:
Hello. You can configure the projector using the TI GUI if lc4500startup does not work. Put it in pattern sequence mode with HDMI input. Set your graphics card to the native (diamond pixel) 914x1140 resolution. Please read our paper on calibration for an explanation.

nikkadim, 2016-11-16:
Thanks @jakobwilm for the tip with _native_ resolution! 

That is really strange,  I set projector in pattern sequence mode and external (HDMI) pattern input as you  did in lc4500startup, but projector still showing anything and not sending output pulses to camera.

Here is the new SLStudio settings:
[image]

console:
[image]

and the projector settings:
[image]

Thank you in advance.

jakobwilm, 2016-11-16:
Set pattern source to video port and trigger mode to vsync.

closed 2016-11-16 by jakobwilm

nikkadim, 2016-11-16:
It's actually set. TI GUI .3.0.1 for some reason does not show this right.
I've added in lc4500startup.cpp reading procedures (LCR_Get...) just to check the current status after I set LCR_SetPatternDisplayMode and LCR_SetPatternTriggerMode:

>  nikk@STEREO-SYSTEM:~/slstudio/slstudio-master/tools/lc4500startup$ ./lc4500startup
> lc4500startup: preparing LightCrafter 4500 for duty... 
> LCR_GetPatternDisplayMode = 0
> LCR_GetPatternTriggerMode = 0
> done!

so everything looks right..

[image]
[image]

jakobwilm, 2016-11-16:
Well, the firmware and GUI have their quirks and bugs and usually take a few days to work around... The S172 looks a bit strange. Also, Pattern Sequence [Variable Exposure] is the new default mode. Maybe something was fixed in LC4500 api in the firmware 3 release. I would try setting it up using the GUI first in Variable Exposure.