jakobwilm/slstudio #13: License?

nito5191, 2017-03-03:
Hi,

Thanks for sharing an interesting project! I am interested in using parts of this project in a closed source  application. Is that possible? What is the license of slstudio?

Regards,
Nils

jakobwilm, 2017-03-07:
Hi Nils,

thank you for your interest. Currently SLStudio is licence free. As such, the code is covered under copyright. You are welcome to use it for personal use but not distribute it in any way.

As for commercial licensing, I would be open for that, but it would involve negotiation with my employer (DTU Compute, Technical University of Denmark). I am not familiar with the process or the usual terms. Please send me an email if you or your company would be interested.

Regards, 
Jakob Wilm

devernay, 2019-02-12:
Hi Jakob,

You should probably clarify this either in the toplevel README or in a COPYING or LICENSE file.

The about dialog mentions (c) 2014 GPL (and others), so I guess your first intent was to release that as GPL? https://github.com/jakobwilm/slstudio/blob/master/src/SLAboutDialog.ui#L69

Would you or your university be OK to release it under a more liberal license (BSD or MIT), since it is already 5 years old?

Thank you,
Frederic

jakobwilm, 2019-02-13:
Hi Frederic,

you are right. I will clear this issue with the university and make the licensing terms clear.

Regards,
Jakob

jakobwilm, 2021-08-30:
Hello Again,

SLStudio is now licensed under GPLv3 with the option to optain a commercial license.

Sincerely

closed 2022-04-24 by jakobwilm