jakobwilm/slstudio #16: 120Hz

nikkadim, 2017-08-02:
@jakobwilm  @ereh 

In readme file you saying that "_LightCrafter4500 configured for 120Hz HDMI gray-scale projection (e.g. configure using /tools/lc4500startup)_"but indeed lc4500startup does not switch to 120Hz LC4500. Do you have any recommendation how to set 120Hz as output for second screen (Ubuntu)?

Thank you.

jakobwilm, 2017-08-07:
Hello. The program lc4500startup is supposed to configure the projector for "external pattern mode" with exposure time == exposure period == 8333 us, which equals 120Hz. Every pattern in repeated twice in succession and captured once in a 8333us exposure by the camera (some of the first some of the second projector exposure) to enable full capture of the pattern while respecting camera latencies. The included TI API was released with the 2.x firmware of the dlpc350, and has since been updated by TI. Maybe it does not play well with the 3.x firmware.

closed 2017-08-07 by jakobwilm

nikkadim, 2017-08-07:
@jakobwilm  "external pattern mode" - means patterns from HDMI, or from flash? I thought you configured graphic card to output 120Hz out from HDMI, no?

jakobwilm, 2017-08-07:
It means from HDMI. I usually configure for 60Hz from graphics card.