jakobwilm/slstudio #17: Triangulator algorithm and some questions

fangchuan, 2017-08-03:
Hi man, I have studied on SLStudio for some time and now have a few questions to ask you。Could you have time to help me solve these questions?
1、What is the position of the phase algorithm in Triangulaotr? It is neither the traditional fixed position of the phase algorithm, nor is the flexible position of the phase algorithm, I did not understand the algorithm in Triangulator.cpp
2、I have  tested the first six codec algorithms, but the point cloud model shown in the SLPointCloudWidget is still incorrect and can clearly see the shadows left by the black and white stripes on the surface of the object, causing the depth of the object's surface  as a sine change, I think this is caused by the solution is not correct.

Hope to get your reply as soon as possible，thanks!﻿ @

nikkadim, 2017-08-03:
@fangchuan 
2. What calibration error did you get? Should be <1 at least.  Check the gamma correction from source, likely you need to adjust it.  If your HDMI source @60Hz  - projection time will then be t = 1/60 = 16.667 ms. So in that case, the minimum camera exposure required is 16.667 ms. That will make sure the entire projected frame is captured. If longer exposures are required, then make the exposure time an integer multiple of 16.667 ms.

fangchuan, 2017-08-03:
@nikkadim 
Thank you for your reply, according to your argument I have solved the problem of the depth of object's surface, but the output of 4-phaseShift codec is still wrong, the depth of point cloud model  is still the same as the sine curve changes, I checked the code, which is similar to the N-phase shift encoder code , but  N-phaseShift codec's out is right. Even with my own writed getPhaseFor4Phase (), but the output is still the depth of information is also not correct

nikkadim, 2017-08-03:
@fangchuan sorry, did not try 4-phaseShift codec

fangchuan, 2017-08-04:
@nikkadim 
That is the result of my camera's calibration:
<cam_error>2.2046176200225712e-001</cam_error>
<proj_error>3.7952080356169646e+000</proj_error>
<stereo_error>3.9291731193646475e+000</stereo_error>

If i tilt the model plane at a great angle, proj_error will be bigger, is this normal?

nikkadim, 2017-08-04:
@fangchuan 
cam error seems ok, but proj - is bad. The calibration obviously goes wrong. It needs to be fixed before anything else. The "result images" in the calibration dialog are supposed to be "flat illumination images" without  stripes, you probably have it. You either do not have a linear gray-value response or the images are saturated. @jakobwilm has gamma-check Matlab-script to check, I did not try it.

fangchuan, 2017-08-04:
@nikkadim 
yes, every image i get in the calibration dialog have strips. I have modified the camera and projector GAMMA = 2.5, the pictures are brighter. But the pictures are still striped, the calibration results when GAMMA = 2.5: 
Seq err 1: cam = 0.361    proj = 0.8007
Seq err 2: cam = 0.3893  proj = 0.8143
Seq err 3: cam =0.6125   proj = 1.127
Seq err 4: cam = 0.6991  proj = 1.108

Another problem,  the Calibration video on the youtube is just a simple translation of the calibration board, is this effective? In Zhang's calibration paper, at least two images are required in different orientation。

nikkadim, 2017-08-04:
@fangchuan  in readme file @jakobwilm  suggested to perform 20 positions

jakobwilm, 2017-08-04:
Yes, pure translation is not sufficient. The video serves only  demonstration purposes.
If result images are not flat, you probably have a non-linear grayvalue response, which needs to be fixed. Unfortunately, consumer projectors perform all sorts of "image processing", degamma, sharpening, spatiotemporal dithering - all of which are bad for measurement purposes.
Try to generate a curve of the gray-value response using the matlab script and wrappers. If you cant get it linear by setting appropriate values in the projector's and graphics card settings, you could also atempt to implement a calibration correction step. This has been described in literature many times.
You should expect reproj errors below 1px for both camera and projector.
Best regards,
/Jakob

fangchuan, 2017-08-05:
@jakobwilm 
Thank you for your reply. In addition, I have not understand  the position of phase  algorithm in the Triangulator.cpp.Could you  explain to me or tell me the literature for reference?

jakobwilm, 2017-08-07:
Sorry, I don’t understand this question.

fangchuan, 2017-08-07:
@jakobwilm 
I do not understande the algorithm in Triangulator.cpp，What does the "determinantTensor" mean? Why is it calculated like this?

jakobwilm, 2017-08-07:
Those are some pre-calculations on the triangulation equation. Essentially, the system Ax=b is solved for every scene point. Only up (or vp in vertical camera-projector configuration) changes between time points, hence solving can be much accelerated. This was adapted from litterature.

closed 2017-08-07 by jakobwilm