jakobwilm/slstudio #21: Strange calibration and reconstruction results

yujiazhang777, 2017-12-19:
Hi, 

I always got incorrect calibration result. The camera calibration is good but the projector calibration result is totally incorrect, as well as the stereo calibration. The reprojection error of projector or stereo is more than e+02. The point cloud result is just a plane without depth information. 
I have tried to load the correct calibration file, but the 3D result is messed up. I also tried the matlab code and made my gamma response linear. 
Do you know how to resolve this problem?

Bests, 
Yujia

benrennison, 2018-01-18:
Yujia, Did this ever get resolved?

nikkadim, 2018-01-18:
@benrennison  what's in particular? Have no problem with calibration. 
Did you play with gamma for your graphics card?

yujiazhang777, 2018-01-23:
@benrennison Still not resolved. 
@nikkadim I have problem in projector calibration, the reprojection error is large. The camera calibration does not have problem. So I think the problem is in the decoding process. I loaded the correct calibration file in slstudio but the point cloud still incorrect, even using the conventional gray codec. 
The gamma response is not very linear but I tried to change the gamma of my graphics card and the response is much more linear (the figure shows my gamma response 
[image]
). But it still not work.

jakobwilm, 2018-01-23:
You need to get the response linear. Otherwise there is just simply no way it will ever work.

/Jakob

[epson-response-0 9]<https://user-images.githubusercontent.com/32459738/35290342-f690a174-0037-11e8-8ed8-b06fe2053fb5.jpg>

closed 2018-01-31 by jakobwilm

Cynthia7, 2018-03-05:
Hi Jakob,

I have made the gamma response linear like the figure shows. 
[image]
But I still have errors in calibration and the projector and stereo calibration are totally incorrect as shown below:
Camera Calibration: 
- cam_error:
0.6385
- Kc:
[1834.4885, 0, 1279.5;
  0, 1834.4885, 959.5;
  0, 0, 1]
- kc:
[-0.09699, 0, 0, 0, 0]
Projector Calibration: 
- proj_error: 
4.936
- Kp: 
[9959.1504, 0, 1066.7208;
  0, 9959.1504, 275.43365;
  0, 0, 1]
- kp: 
[-4.126, 0, 0, 0, 0]
Stereo Calibration: 
- stereo_error:
10.98
- Rp:
[0.99225891, -0.015919777, 0.12316168;
  -0.099331938, 0.49348894, 0.86406124;
  -0.074534588, -0.86960638, 0.48808745]
- Tp:
[-70.22, -1080, 4685]
Seq error 1 cam:0.4911 proj:1.997
Seq error 2 cam:0.359 proj:3.491
Seq error 3 cam:0.6174 proj:7.768
Seq error 4 cam:0.4521 proj:4.641
Seq error 5 cam:0.751 proj:3.172
Seq error 6 cam:0.328 proj:3.354
Seq error 7 cam:0.6916 proj:3.831
My captured image is like this:
[image]
Is this because that my camera image is too dark since the gamma response shows that I can only capture intensity about 120 when the projected intensity value is 255?

jakobwilm, 2018-03-05:
Hi Cynthia,

Your data looks alright. But you should make sure that tje algorithm does not use (very noisy) data from the black checket fields. Try to write the occlusion mask to disk and/or print a grey on white checkerboard.

/Jakob

[gamma_2 1_dark_03042018]<https://user-images.githubusercontent.com/3238223/36986891-0148a75a-2069-11e8-9d34-bc40a5b7f32a.jpg>

[frameseq_02]<https://user-images.githubusercontent.com/3238223/36987323-0d242198-206a-11e8-8682-87f5934fe5a8.jpg>

Cynthia7, 2018-03-05:
OK, I will try that. Thank you for your suggestion.

Bests,
Yujia Zhang