jakobwilm/slstudio #22: Error running latest branch

DPhysicsBris, 2018-02-23:
Using ubuntu 16.04 64-bit and the latest branch of slstudio, I am encountering these two errors when running the calibration. The error occurs shortly after opening the window, without using any of the buttons or options in the calibration window. 

Using earlier branches this error did not occur. The second error here seems to be something in the code, rather than our hardware, however I am unsure on the openCV error. 

Error messages:

```
...
CameraXIMEA: Error at xiGetImage (10)
OpenCV Error: Assertion failed (s >= 0) in setSize, file /build/opencv-SviWsf/opencv-2.4.9.1+dfsg/modules/core/src/matrix.cpp, line 116
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-SviWsf/opencv-2.4.9.1+dfsg/modules/core/src/matrix.cpp:116: error: (-215) s >= 0 in function setSize
```

```
...
CameraXIMEA: Error at xiGetImage (10)
OpenCV Error: Insufficient memory (Failed to allocate 23258632277584 bytes) in OutOfMemoryError, file /build/opencv-SviWsf/opencv-2.4.9.1+dfsg/modules/core/src/alloc.cpp, line 52
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-SviWsf/opencv-2.4.9.1+dfsg/modules/core/src/alloc.cpp:52: error: (-4) Failed to allocate 23258632277584 bytes in function OutOfMemoryError
```

Hardware:
DLP lightcrafter 3500
Ximea Xiq MQ013MG-E2

jakobwilm, 2018-02-23:
Hi DPhysicsBris, your Ximea camera does probably timeout in acquisition. Maybe it is not receiveing the hardware trigger properly? Try building in debug and run with debugger to find the cause.

closed 2018-02-23 by jakobwilm

DPhysicsBris, 2018-02-28:
Hi jakobwilm,

Thank you for getting back to us so quickly. 

The second error listed there is given when the Ximea parameters are changed on lines 78 and 80 of CameraXIMEA.cpp from 1024 and 512 to 1280 and 1024 (the dimensions of our camera). We also adjusted the offset parameters on lines 82 and 84 of the same file but that made no difference. 

Are there any other lines that would need to be adjusted to change the region of interest of the camera feed?

When I leave the width and height parameters as the default 1024 and 512 I do not get the getimage error. 

Thanks
Chris