jakobwilm/slstudio #24: Problem with 3D reconstruction

meardon, 2018-03-19:
Hi Jakob,

Upon calibration, we receive the following results. Then using this calibration and the attached preferences, we get a 3D reconstruction that looks like the other image attached. 

It appears to us that the object shown in the 3D reconstruction should be at the crossing of the axis lines, but is instead displaced some amount rendering it hard to view.

We are also under the impression that the feint 3D reconstruction shown is not, in fact, a 3D reconstruction, but instead just our camera frames, which seems bizarre. 

Do you have any ideas or resolutions on how we can fix this?

We are using ubuntu 16.04 64-bit with the latest branch of slstudio, with a minor change to use pcl 1.7 instead of 1.8 (changing the version in the slstudio.pro file). Is this version change the issue? 

Hardware:
DLP lightcrafter 3500 (with a modification to output a 5V trigger signal by moving some resistors on its PCB)
Ximea Xiq MQ013MG-E2

[image]

[image]
`========================================
Camera Calibration: 
- cam_error:
0.8049
- Kc:
[19950.707, 0, 639.5;
  0, 19950.707, 271.5;
  0, 0, 1]
- kc:
[17.54, 0, 0, 0, 0]
Projector Calibration: 
- proj_error: 
96.3
- Kp: 
[9287.7764, 0, -2014.4105;
  0, 9287.7764, 678.02643;
  0, 0, 1]
- kp: 
[-1.55, 0, 0, 0, 0]
Stereo Calibration: 
- stereo_error:
75.11
- Rp:
[0.958996, 0.056349218, 0.27776149;
  0.068554565, 0.9048152, -0.42024967;
  -0.27500355, 0.42205957, 0.86385113]
- Tp:
[-479.1, 1337, -1775]
Seq error 1 cam:0.6059 proj:30.7
Seq error 2 cam:0.6556 proj:72.49
Seq error 3 cam:0.4669 proj:68.86
Seq error 4 cam:0.5649 proj:44.43
Seq error 5 cam:0.3978 proj:72.67
Seq error 6 cam:0.9967 proj:75.99
Seq error 7 cam:0.4299 proj:60.53
Seq error 8 cam:0.553 proj:34.98
Seq error 9 cam:0.6184 proj:65.53
Seq error 10 cam:0.4295 proj:76.21
Seq error 11 cam:0.4578 proj:66.2
Seq error 12 cam:0.5199 proj:44.22
Seq error 13 cam:0.6177 proj:74.11
Seq error 14 cam:0.686 proj:82.97
Seq error 15 cam:0.4333 proj:97.14
Seq error 16 cam:0.7851 proj:68.74
Seq error 17 cam:0.5311 proj:95.8
Seq error 18 cam:0.5311 proj:39.25
Seq error 19 cam:0.3853 proj:55
Seq error 20 cam:1.188 proj:151.4
Seq error 21 cam:0.7961 proj:95.36
Seq error 22 cam:0.4313 proj:62.24
Seq error 23 cam:0.5 proj:54.2
Seq error 24 cam:0.5939 proj:88.12
Seq error 25 cam:0.5404 proj:60.88
`
[image]

[image]

jakobwilm, 2018-03-19:
Hello.

The projector calibration fails. You must definately fix thia before you can expect any reasonable results.
Your delay parameter in the settings seema very short. Most projectors have considerable input latency.
After fixing that, please make sure that the projector-camera response is linear, as discussed in several previous messages.
Good luck!

/Jakob

[screenshot from 2018-03-19 16-03-52]<https://user-images.githubusercontent.com/34161309/37607224-5fd63086-2b8f-11e8-916f-10bdf48eebbb.png>

[screenshot from 2018-03-19 15-51-32]<https://user-images.githubusercontent.com/34161309/37607049-e850f53c-2b8e-11e8-99e0-7cec9a0fa36c.png>
`========================================

[img_0175]<https://user-images.githubusercontent.com/34161309/37607040-e0ee76e8-2b8e-11e8-9ec3-2a6b0d078357.jpg>

[screenshot from 2018-03-19 15-52-41]<https://user-images.githubusercontent.com/34161309/37607046-e4999322-2b8e-11e8-9f15-209ad09ef6a0.png>

nikkadim, 2018-03-19:
@meardon tune gamma correction on GPU to get rid of stripes on calibration pimages.

meardon, 2018-03-21:
Hi Jakob, 

Thank you very much for your response. We have increased our delay setting to: 100ms. 

We have also found that updating a graphics driver has improved our calibrations but a large amount.

However, we are trying to use the matlab script that you provided and we are having difficulties.
Do we need to change any of the code in the matlab files to get any results, or should we just be running a particular file from the folder within matlab? 

We have run determinegammaresponse and we are getting the error:
`Warning: The following error was caught while executing 'Projector' class destructor:
Undefined function 'ProjectorMex' for input arguments of type 'char'.

Error in Projector/delete (line 24)
            ProjectorMex('delete', this.objectHandle);

Error in Projector (line 18)
        function this = Projector(screenNum)

Error in determineGammaResponse (line 5)
p = Projector(1); 
> In Projector (line 18)
  In determineGammaResponse (line 5) 
Undefined function or variable 'ProjectorMex'.

Error in Projector (line 19)
            this.objectHandle = ProjectorMex('new', screenNum);

Error in determineGammaResponse (line 5)
p = Projector(1);`

@nikkadim we have found how to tune the gamma response we are just not currently able to measure it after the processing of the camera and projector. 

Thanks,
Chris

jakobwilm, 2018-03-21:
Hi Chris,

You will need to compile the mex-functions, which wrap C++ into Matlab callable functions.
The make.m script should handle that, but there may be adjustments necessary depending on your compiler, library versions, etc.

Jakob Wilm
Postdoctoral Researcher
DTU Compute
---------------------------------------------------
Technical University of Denmark
Department of Applied Mathematics and Computer Science
Richard Petersens Plads
Building 321/220
2800 Lyngby
jakw@dtu.dk<mailto:jakw@dtu.dk>
+45 45 25 37 16
http://www.compute.dtu.dk

meardon, 2018-03-21:
Hi Jakob, 

I apologise if I am being stupid here, I am new to matlab. Using matlab version 9.4.0.813654 (R2018a) we are getting a brace indexing error on line 32 of the make.m file. This is the line where the mex function is called. The exact error message is "Brace indexing is not supported for variables of this type" and then quoting line 32. 

Is there something that we need to change or possibly is our matlab version incompatible? 

Thanks
Chris

jakobwilm, 2018-03-21:
Hi Chris,

the syntax was indeed incompatible with the newer/st Matlab. I just pushed a few changes which should fix it.

closed 2018-03-21 by jakobwilm