jakobwilm/slstudio #25: Matlab code - Error when running mex files.

meardon, 2018-03-22:
Hi Jakob,

Using the updated version of the matlab code, we have seemingly successfully built the executables (console output below). Then when running the .mexa64 files in matlab are getting the errors: 

```
>> ProjectorMex
Error using ProjectorMex
First input should be a command string less than 64 characters long.
```
and
```
>> CameraMex
Error using CameraMex
First input should be a command string less than 64 characters long.
```

This seems as if arguments are being expected? To run each of these .mexa64 files we are calling ProjectorMex or CameraMex in the console without any parentheses. Are we doing this incorrectly?

Gcc version: 6.3.0
Ubuntu 16.04 64-bit
matlab R2018a

Build output:
```
>> make
Verbose mode is on.
... Looking for compiler 'g++' ...
... Executing command 'which g++' ...Yes ('/usr/bin/g++').
... Executing command 'g++ -print-file-name=libstdc++.so' ...Yes ('/usr/lib/gcc/x86_64-linux-gnu/6/libstdc++.so').
... Executing command 'g++ -dumpversion' ...Yes ('6.3.0').
... Executing command 'which g++' ...Yes ('/usr/bin/g++').
... Looking for folder '/usr' ...Yes.
... Executing command 'g++ -dumpmachine' ...Yes ('x86_64-linux-gnu').
Found installed compiler 'g++'.
Set INCLUDE = /usr/lib/gcc/x86_64-linux-gnu/6.3.0/include;/usr/include/c++/6.3.0;/usr/include/c++/6.3.0/x86_64-linux-gnu;/usr/include/c++/6.3.0/backward;/usr/lib/gcc/x86_64-linux-gnu/6.3.0/include;/usr/include/c++/6.3.0;/usr/include/c++/6.3.0/x86_64-linux-gnu;/usr/include/c++/6.3.0/backward;
Options file details

	Compiler location: /usr/bin/g++
	Options file: /usr/local/MATLAB/R2018a/bin/glnxa64/mexopts/g++_glnxa64.xml
	CMDLINE2 : /usr/bin/g++ -pthread -Wl,--no-undefined  -shared -O -Wl,--version-script,"/usr/local/MATLAB/R2018a/extern/lib/glnxa64/c_exportsmexfileversion.map" /tmp/mex_975137339462_1885/CameraMex.o /tmp/mex_975137339462_1885/Camera.o /tmp/mex_975137339462_1885/CameraIIDC.o /tmp/mex_975137339462_1885/CameraXIMEA.o /tmp/mex_975137339462_1885/cpp_mexapi_version.o   -ldc1394  -lm3api   -Wl,--as-needed -Wl,-rpath-link,/usr/local/MATLAB/R2018a/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/bin/glnxa64" -Wl,-rpath-link,/usr/local/MATLAB/R2018a/extern/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/extern/bin/glnxa64" -lMatlabDataArray -lmx -lmex -lmat -lm -lstdc++ -o CameraMex.mexa64
	CXX : /usr/bin/g++
	DEFINES : -DWITH_CAMERAIIDC  -DWITH_CAMERAXIMEA  -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE 
	MATLABMEX : -DMATLAB_MEX_FILE 
	CFLAGS : -fexceptions -fPIC -fno-omit-frame-pointer -pthread
	CXXFLAGS : -fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11
	INCLUDE : -I"../src/camera" -I"/usr/local/lib" -I"/opt/XIMEA/include"  -I"/usr/local/MATLAB/R2018a/extern/include" -I"/usr/local/MATLAB/R2018a/simulink/include"
	CXXOPTIMFLAGS : -O -DNDEBUG
	CXXDEBUGFLAGS : -g
	LDXX : /usr/bin/g++
	LDFLAGS : -pthread -Wl,--no-undefined 
	LDTYPE : -shared
	LINKEXPORT : -Wl,--version-script,"/usr/local/MATLAB/R2018a/extern/lib/glnxa64/mexFunction.map"
	LINKEXPORTVER : -Wl,--version-script,"/usr/local/MATLAB/R2018a/extern/lib/glnxa64/c_exportsmexfileversion.map"
	LINKLIBS : -ldc1394  -lm3api   -Wl,--as-needed -Wl,-rpath-link,/usr/local/MATLAB/R2018a/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/bin/glnxa64" -Wl,-rpath-link,/usr/local/MATLAB/R2018a/extern/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/extern/bin/glnxa64" -lMatlabDataArray -lmx -lmex -lmat -lm -lstdc++
	LDOPTIMFLAGS : -O
	LDDEBUGFLAGS : -g
	MWCPPLIB : "/usr/local/MATLAB/R2018a/sys/os/glnxa64/libstdc++.so.6"
	OBJEXT : .o
	LDEXT : .mexa64
	SETENV : CC="gcc"
                CXX="/usr/bin/g++"
                CFLAGS="-fexceptions -fPIC -fno-omit-frame-pointer -pthread -DWITH_CAMERAIIDC  -DWITH_CAMERAXIMEA  -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE "
                CXXFLAGS="-fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11 -DWITH_CAMERAIIDC  -DWITH_CAMERAXIMEA  -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE "
                COPTIMFLAGS="-O -DNDEBUG"
                CXXOPTIMFLAGS="-O -DNDEBUG"
                CDEBUGFLAGS="-g"
                CXXDEBUGFLAGS="-g"
                LD="gcc"
                LDXX="/usr/bin/g++"
                LDFLAGS="-pthread -Wl,--no-undefined  -shared -ldc1394  -lm3api   -Wl,--as-needed -Wl,-rpath-link,/usr/local/MATLAB/R2018a/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/bin/glnxa64" -Wl,-rpath-link,/usr/local/MATLAB/R2018a/extern/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/extern/bin/glnxa64" -lMatlabDataArray -lmx -lmex -lmat -lm -lstdc++ -Wl,--version-script,"/usr/local/MATLAB/R2018a/extern/lib/glnxa64/mexFunction.map""
[... 100 lines cut ...]
	EXE : ProjectorMex.mexa64
	ILK : "ProjectorMex.ilk"
	MANIFEST : "ProjectorMex.mexa64.manifest"
	TEMPNAME : ProjectorMex
	EXEDIR : 
	EXENAME : ProjectorMex
	OPTIM : -O -DNDEBUG
	LINKOPTIM : -O
	CMDLINE1_0 : /usr/bin/g++ -c -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I"../src/projector"  -I"/usr/local/MATLAB/R2018a/extern/include" -I"/usr/local/MATLAB/R2018a/simulink/include" -fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11 -O -DNDEBUG "/home/physics/project/slstudio/matlab/ProjectorMex.cpp" -o /tmp/mex_982710491912_1885/ProjectorMex.o
	CMDLINE1_1 : /usr/bin/g++ -c -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I"../src/projector"  -I"/usr/local/MATLAB/R2018a/extern/include" -I"/usr/local/MATLAB/R2018a/simulink/include" -fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11 -O -DNDEBUG "/home/physics/project/slstudio/matlab/../src/projector/ProjectorOpenGL.cpp" -o /tmp/mex_982710491912_1885/ProjectorOpenGL.o
	CMDLINE1_2 : /usr/bin/g++ -c -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I"../src/projector"  -I"/usr/local/MATLAB/R2018a/extern/include" -I"/usr/local/MATLAB/R2018a/simulink/include" -fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11 -O -DNDEBUG "/home/physics/project/slstudio/matlab/../src/projector/OpenGLContext.Unix.cpp" -o /tmp/mex_982710491912_1885/OpenGLContext.Unix.o
	CMDLINE1_3 : /usr/bin/g++ -c -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I"../src/projector"  -I"/usr/local/MATLAB/R2018a/extern/include" -I"/usr/local/MATLAB/R2018a/simulink/include" -fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11 -O -DNDEBUG "/usr/local/MATLAB/R2018a/extern/version/cpp_mexapi_version.cpp" -o /tmp/mex_982710491912_1885/cpp_mexapi_version.o

Building with 'g++'.
/usr/bin/g++ -c -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I"../src/projector"  -I"/usr/local/MATLAB/R2018a/extern/include" -I"/usr/local/MATLAB/R2018a/simulink/include" -fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11 -O -DNDEBUG "/home/physics/project/slstudio/matlab/ProjectorMex.cpp" -o /tmp/mex_982710491912_1885/ProjectorMex.o
/usr/bin/g++ -c -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I"../src/projector"  -I"/usr/local/MATLAB/R2018a/extern/include" -I"/usr/local/MATLAB/R2018a/simulink/include" -fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11 -O -DNDEBUG "/home/physics/project/slstudio/matlab/../src/projector/ProjectorOpenGL.cpp" -o /tmp/mex_982710491912_1885/ProjectorOpenGL.o
/usr/bin/g++ -c -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I"../src/projector"  -I"/usr/local/MATLAB/R2018a/extern/include" -I"/usr/local/MATLAB/R2018a/simulink/include" -fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11 -O -DNDEBUG "/home/physics/project/slstudio/matlab/../src/projector/OpenGLContext.Unix.cpp" -o /tmp/mex_982710491912_1885/OpenGLContext.Unix.o
/usr/bin/g++ -c -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I"../src/projector"  -I"/usr/local/MATLAB/R2018a/extern/include" -I"/usr/local/MATLAB/R2018a/simulink/include" -fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11 -O -DNDEBUG "/usr/local/MATLAB/R2018a/extern/version/cpp_mexapi_version.cpp" -o /tmp/mex_982710491912_1885/cpp_mexapi_version.o
/usr/bin/g++ -pthread -Wl,--no-undefined  -shared -O -Wl,--version-script,"/usr/local/MATLAB/R2018a/extern/lib/glnxa64/c_exportsmexfileversion.map" /tmp/mex_982710491912_1885/ProjectorMex.o /tmp/mex_982710491912_1885/ProjectorOpenGL.o /tmp/mex_982710491912_1885/OpenGLContext.Unix.o /tmp/mex_982710491912_1885/cpp [... cut ...]
/usr/bin/g++ -pthread -Wl,--no-undefined  -shared -O -Wl,--version-script,"/usr/local/MATLAB/R2018a/extern/lib/glnxa64/c_exportsmexfileversion.map" /tmp/mex_982710491912_1885/ProjectorMex.o /tmp/mex_982710491912_1885/ProjectorOpenGL.o /tmp/mex_982710491912_1885/OpenGLContext.Unix.o /tmp/mex_982710491912_1885/cpp_mexapi_version.o   -lGL  -lGLU  -lX11  -lXxf86vm  -lGLEW   -Wl,--as-needed -Wl,-rpath-link,/usr/local/MATLAB/R2018a/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/bin/glnxa64" -Wl,-rpath-link,/usr/local/MATLAB/R2018a/extern/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/extern/bin/glnxa64" -lMatlabDataArray -lmx -lmex -lmat -lm -lstdc++ -o ProjectorMex.mexa64
MEX completed successfully.
>> 
```

jakobwilm, 2018-03-22:
Yes, please look at the gamma-script to learn how to use these wrappers.

>> ProjectorMex
Error using ProjectorMex
First input should be a command string less than 64 characters long.

and

>> CameraMex
Error using CameraMex
First input should be a command string less than 64 characters long.

This seems as if arguments are being expected? To run each of these .mexa64 files we are calling ProjectorMex or CameraMex in the console without any parentheses. Are we doing this incorrectly?

Gcc version: 6.3.0
Ubuntu 16.04 64-bit
matlab R2018a

Build output:

>> make
Verbose mode is on.
... Looking for compiler 'g++' ...
... Executing command 'which g++' ...Yes ('/usr/bin/g++').
... Executing command 'g++ -print-file-name=libstdc++.so' ...Yes ('/usr/lib/gcc/x86_64-linux-gnu/6/libstdc++.so').
... Executing command 'g++ -dumpversion' ...Yes ('6.3.0').
... Executing command 'which g++' ...Yes ('/usr/bin/g++').
... Looking for folder '/usr' ...Yes.
... Executing command 'g++ -dumpmachine' ...Yes ('x86_64-linux-gnu').
Found installed compiler 'g++'.
Set INCLUDE = /usr/lib/gcc/x86_64-linux-gnu/6.3.0/include;/usr/include/c++/6.3.0;/usr/include/c++/6.3.0/x86_64-linux-gnu;/usr/include/c++/6.3.0/backward;/usr/lib/gcc/x86_64-linux-gnu/6.3.0/include;/usr/include/c++/6.3.0;/usr/include/c++/6.3.0/x86_64-linux-gnu;/usr/include/c++/6.3.0/backward;
Options file details

        Compiler location: /usr/bin/g++
        Options file: /usr/local/MATLAB/R2018a/bin/glnxa64/mexopts/g++_glnxa64.xml
        CMDLINE2 : /usr/bin/g++ -pthread -Wl,--no-undefined  -shared -O -Wl,--version-script,"/usr/local/MATLAB/R2018a/extern/lib/glnxa64/c_exportsmexfileversion.map" /tmp/mex_975137339462_1885/CameraMex.o /tmp/mex_975137339462_1885/Camera.o /tmp/mex_975137339462_1885/CameraIIDC.o /tmp/mex_975137339462_1885/CameraXIMEA.o /tmp/mex_975137339462_1885/cpp_mexapi_version.o   -ldc1394  -lm3api   -Wl,--as-needed -Wl,-rpath-link,/usr/local/MATLAB/R2018a/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/bin/glnxa64" -Wl,-rpath-link,/usr/local/MATLAB/R2018a/extern/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/extern/bin/glnxa64" -lMatlabDataArray -lmx -lmex -lmat -lm -lstdc++ -o CameraMex.mexa64
        CXX : /usr/bin/g++
        DEFINES : -DWITH_CAMERAIIDC  -DWITH_CAMERAXIMEA  -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE
        MATLABMEX : -DMATLAB_MEX_FILE
        CFLAGS : -fexceptions -fPIC -fno-omit-frame-pointer -pthread
        CXXFLAGS : -fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11
        INCLUDE : -I"../src/camera" -I"/usr/local/lib" -I"/opt/XIMEA/include"  -I"/usr/local/MATLAB/R2018a/extern/include" -I"/usr/local/MATLAB/R2018a/simulink/include"
        CXXOPTIMFLAGS : -O -DNDEBUG
        CXXDEBUGFLAGS : -g
        LDXX : /usr/bin/g++
        LDFLAGS : -pthread -Wl,--no-undefined
        LDTYPE : -shared
        LINKEXPORT : -Wl,--version-script,"/usr/local/MATLAB/R2018a/extern/lib/glnxa64/mexFunction.map"
        LINKEXPORTVER : -Wl,--version-script,"/usr/local/MATLAB/R2018a/extern/lib/glnxa64/c_exportsmexfileversion.map"
        LINKLIBS : -ldc1394  -lm3api   -Wl,--as-needed -Wl,-rpath-link,/usr/local/MATLAB/R2018a/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/bin/glnxa64" -Wl,-rpath-link,/usr/local/MATLAB/R2018a/extern/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/extern/bin/glnxa64" -lMatlabDataArray -lmx -lmex -lmat -lm -lstdc++
        LDOPTIMFLAGS : -O
        LDDEBUGFLAGS : -g
        MWCPPLIB : "/usr/local/MATLAB/R2018a/sys/os/glnxa64/libstdc++.so.6"
        OBJEXT : .o
        LDEXT : .mexa64
        SETENV : CC="gcc"
                CXX="/usr/bin/g++"
                CFLAGS="-fexceptions -fPIC -fno-omit-frame-pointer -pthread -DWITH_CAMERAIIDC  -DWITH_CAMERAXIMEA  -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE "
                CXXFLAGS="-fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11 -DWITH_CAMERAIIDC  -DWITH_CAMERAXIMEA  -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE "
                COPTIMFLAGS="-O -DNDEBUG"
                CXXOPTIMFLAGS="-O -DNDEBUG"
                CDEBUGFLAGS="-g"
                CXXDEBUGFLAGS="-g"
                LD="gcc"
                LDXX="/usr/bin/g++"
                LDFLAGS="-pthread -Wl,--no-undefined  -shared -ldc1394  -lm3api   -Wl,--as-needed -Wl,-rpath-link,/usr/local/MATLAB/R2018a/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/bin/glnxa64" -Wl,-rpath-link,/usr/local/MATLAB/R2018a/extern/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/extern/bin/glnxa64" -lMatlabDataArray -lmx -lmex -lmat -lm -lstdc++ -Wl,--version-script,"/usr/local/MATLAB/R2018a/extern/lib/glnxa64/mexFunction.map""
                LDDEBUGFLAGS="-g"
        GCC : /usr/bin/g++
        CPPLIB_DIR : /usr/lib/gcc/x86_64-linux-gnu/6/libstdc++.so
        VER : 6.3.0
[... 94 lines cut ...]
        EXP : "ProjectorMex.exp"
        LIB : "ProjectorMex.lib"
        EXE : ProjectorMex.mexa64
        ILK : "ProjectorMex.ilk"
        MANIFEST : "ProjectorMex.mexa64.manifest"
        TEMPNAME : ProjectorMex
        EXEDIR :
        EXENAME : ProjectorMex
        OPTIM : -O -DNDEBUG
        LINKOPTIM : -O
        CMDLINE1_0 : /usr/bin/g++ -c -DMATLAB_DEFAULT_RELEASE=R2017b  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I"../src/projector"  -I"/usr/local/MATLAB/R2018a/extern/include" -I"/usr/local/MATLAB/R2018a/simulink/include" -fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11 -O -DNDEBUG "/home/physics/project/slstudio/matla [... cut ...]
/usr/bin/g++ -pthread -Wl,--no-undefined  -shared -O -Wl,--version-script,"/usr/local/MATLAB/R2018a/extern/lib/glnxa64/c_exportsmexfileversion.map" /tmp/mex_982710491912_1885/ProjectorMex.o /tmp/mex_982710491912_1885/ProjectorOpenGL.o /tmp/mex_982710491912_1885/OpenGLContext.Unix.o /tmp/mex_982710491912_1885/cpp_mexapi_version.o   -lGL  -lGLU  -lX11  -lXxf86vm  -lGLEW   -Wl,--as-needed -Wl,-rpath-link,/usr/local/MATLAB/R2018a/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/bin/glnxa64" -Wl,-rpath-link,/usr/local/MATLAB/R2018a/extern/bin/glnxa64 -L"/usr/local/MATLAB/R2018a/extern/bin/glnxa64" -lMatlabDataArray -lmx -lmex -lmat -lm -lstdc++ -o ProjectorMex.mexa64
MEX completed successfully.

meardon, 2018-03-22:
is that the "determineGammaResponse.m" script?

closed 2018-03-22 by jakobwilm