jakobwilm/slstudio #32: Sample calibration file and images to test 2 x 3 phase

johntraynor, 2018-10-12:
Hi Jakob,

Would you have a sample calib.xml and associated 2 x 3 phase set of images I could test with.

I am having trouble balancing getting a good calibration with my setup. I can see stripes in the calibration images so I know it's not right. Whenever I change the video driver settings to get better values for the camera & calibration errors by removing the stripes it compromises the phase patterns produced when actually scanning an object. Having a reference sample calib and images might help me figure out what is going on

Thanks

jakobwilm, 2018-10-14:
Hi johntraynor,

I can possibly help you, but need to know your name, affiliation and hardware.

Regards, 
Jakob

closed 2018-10-14 by jakobwilm

johntraynor, 2018-10-15:
Hi Jakob,

I have a 4500 evm unit for testing. I have been using video mode up to now for testing. I'm trying to get my head around how to load the fringe patterns onto the unit so that I don't have to worry about gamma corrections but I will have to remap the images for diamond configuration if I go this route. Maybe best to contact you offline??