jakobwilm/slstudio #33: Some questions about the decode algorithm named DecoderPhaseShift2x3!!

zhuangleilei, 2018-11-27:
Hello, thank you for your work about the SLStudio. I have a question about the DecoderPhaseShift2x3 class in the SLStudio. Thank you for your answer. This decode algorithm uses single frequency phase shifting images (3 images) and  phase cue patterns(3 images) to get the unwrapped absolute phase image using. In my work, I used three different frequency phase shifting patterns at least to get the absolute phase image. So, could you tell me the theory about DecoderPhaseShift2x3 or just give me the paper name you refered?
thank you very much. @jakobwilm @snesgaard @ereh

zhuangleilei, 2018-11-28:
Maybe giving the paper name in every decoding algorithm class should be a better action for understanding your work? Thanks a lot for your contribution.

jakobwilm, 2018-11-28:
Hi Zhuang,
thanks for your feedback. I do not have a reference for the exact algorithm. If you are familiar with phase shifting, you find that many variants are possible. Essentially, in the 2x3 codec, we are projecting an equivalent phase directly. Utilizing the heterodyne principle has some (theoretical) advantages, but I have not come to implementing it. A good reference is [1].

Best regards, 
Jakob

[1]: Reich Carsten, Ritter Reinhold & Thesing Jan, 1997. 'White light heterodyne principle for 3D- measurement', Proceedings of SPIE Vol. 3100: Sensors, Sensor Systems, and Sensor Data Processing.

jakobwilm, 2018-11-28:
And yes. If I get the time, I will add references in the source files.

closed 2018-11-28 by jakobwilm

zhuangleilei, 2018-11-28:
I got it, and many thanks for your reply again.