jakobwilm/slstudio #34: beginner guide

Jorgininho, 2019-02-22:
Hello,

I'm interested to use your software but I'm quite a beginner in almost all the techno you are using.
Tried to follow the install instruction on ubuntu and get stuck in Qtcreator, when I load your project and compiling it it says a package is missing 
pcl_visualisation-1.8
is there any page which try to explain with more details how slstudio works ? 
like the classic basic workflow to render a 3D image ? (open Qt creator -> open Slstudio project -> compile program)
Thanks

Xephen20, 2022-04-07:
Hi,
How did you solve this issue?

jakobwilm, 2022-04-20:
Hi. I have updated dependencies and instructions to work with Ubuntu 20.04 out of the box. There is also new example data to get you started and let you try reconstruction even without hardware.

closed 2022-04-20 by jakobwilm