jakobwilm/slstudio #35: Wave in point cloud

DotZ1, 2019-02-24:
Hi,

I managed to get the software running on windows but I am seeing a wave in the 3D point cloud. 

I have attached a picture below. Any idea what this is? 

The point cloud itself if of a flat piece of paper and was scanned using the 2 x 3 patterns in one direction (vertical patterns). When I look at your point cloud that in in your video of the face it is very clean. What is going wrong do you think with my point cloud. The problem is very repetitive.

I am using the lightcrafter 4500 in the setup and the calibrations is less than 1 for both the camera and projector. 

[image]

Thanks in advance

Dot

DuanMH, 2019-02-26:
Hello, same problems exist in my experience, such as lightcrafter 4500 cannot project the designed patterns.

nikkadim, 2019-02-26:
If you project patterns from seconds screen - adjust gamma correction.

jakobwilm, 2019-02-26:
This is most probably the same problem as in most issue requests: your projector-camera pair does not have a linear gray-value response. This can have many causes, but you would start by measuring the response to start finding the bug.

closed 2019-02-26 by jakobwilm

DotZ1, 2019-03-04:
Thanks for the feedback. As a test I embedded the patterns straight onto the 4500  and captured them through the camera but I am still getting the waves in my point clouds. Does that mean to might point to a gamma problem in the camera setup as from what I understand the 4500 does not apply any gamma correction to the patterns (it would do this only in video mode on the 4500)?

jakobwilm, 2019-03-04:
Yes, something non-linear in the camera or a trigger timing issue. As I said before: you have to do measurements.

DotZ1, 2019-03-04:
Many thanks for the quick response.

DotZ1, 2019-03-04:
Sorry one last question on the theory of figuring out the gamma response. I understand you have a matlab function that will tell you the response but could you explain the theory a little. Do you just project a fixed set of gray values from the projector (for example all a value of 100 and check the response) or do they go from dark to bright across the width of the image or some other variation. Thanks in advance

matBBA, 2021-09-15:
Hi DotZ1,
This may be closed, but I had similar waves in my application. It turns out the culprit was then neon light in the room. The projector I used projects sinusoidal patterns by concatenating various black & white pictures in a short period. And I guess each b&w pattern was a bit modified by neon light at this exact moment ( as I changes at ~100Hz if you're In Europe).

As soon as I turned the lights off, this effect was gone...