jakobwilm/slstudio #43: how to handle the dynamic scene reconstruction

nashdingsheng, 2021-03-02:
i am a postgratude student from ningbo university china, i have used this project to generate point cloud in static scene,  but my project is how to use structure light to reconstruct a motion entity.  i do not know whether this project  is also suitablefor dynamic scene or not  and theoretically motion compensation should be considered but i did not observe the related code for motion compensation in this project.  simultaneously  i notice the video for this project seems to work well  for generating point clouds of a  moving head?  so i am confused.   could you please give me some suggestion on this issue and   wonder if there are some materials or code  available for structured light reconstruction in dynamic scene,  looking forward to get your reply, thank you very much.

rocketflyer1997, 2021-12-12:
Hello, i'm also a student from China, i'm building a Dynamic 3D Reconstruction project and i can provide some code for you. I also want to ask some questions, can you help me?  Wechat:18122071769

jakobwilm, 2022-04-24:
Hello. I cannot offer free personal consultancy for you. Generally, dynamic scenes can be reconstructed but projection time needs to be kept very short. You will need hardware triggering.

closed 2022-04-24 by jakobwilm