jakobwilm/slstudio #50: About calibrate

MikeDai2020, 2022-05-20:
Thanks for your works.
I test calibate using the example data,  I got:
Camera Calibration: 
- cam_error:
0.3244
- Kc:
[2204.1396, 0, 843.5;
 0, 2204.1396, 675.5;
 0, 0, 1]
- kc:
[-0.05676, 0, 0, 0, 0]
Projector Calibration: 
- proj_error: 
2.576
- Kp: 
[9329.2891, 0, 874.30811;
 0, 9329.2891, 1442.9962;
 0, 0, 1]
- kp: 
[3.604, 0, 0, 0, 0]
Stereo Calibration: 
- stereo_error:
20.48
- Rp:
[0.88852656, -0.11886059, 0.44316214;
 0.13676967, 0.99056607, -0.0085391086;
 -0.43796644, 0.068198368, 0.89640081]
- Tp:
[-429.7, -208.6, 1961]
**maybe something wrong with the Projector Calibration and Stereo Calibration.**
my setting is following:
chechersize = 15
checkerrows = 9;
checkcols = 13;

my edit code:

`std::vector<cv::Mat> readImages(std::string path)
{
    std::vector<cv::Mat> images;
    std::vector<std::string> names;
    cv::glob(path, names);
    for(auto name: names)
    {
        std::cout << name << std::endl;
        cv::Mat mat = cv::imread(name, cv::IMREAD_UNCHANGED);
        images.push_back(mat);
    }
    return images;
}

void SLCalibrationDialog::on_calibrateButton_clicked() {

  // Disable interface elements
  ui->calibrateButton->setEnabled(false);
  ui->listWidget->setEnabled(false);

  // Stop live view
  killTimer(liveViewTimer);
  reviewMode = true;
  ui->snapButton->setText("Live View");

  calibrator->reset();

  // Note which frame sequences are used
  activeFrameSeqs.clear();

  QListWidget* widget = ui->listWidget;
  for (int i = 0; i < widget->count(); i++) {
     QListWidgetItem* item = widget->item(i);
    if (item->checkState() == Qt::Checked) {
//      vector<cv::Mat> frameSeq(frameSeqs[i].begin(),
//                               frameSeqs[i].begin() +
//                                   calibrator->getNPatterns());
      std::string path = rootPath + "/" + item->text().toStdString();
      std::cout << "add path:\n" << path << std::endl;
      std::vector<cv::Mat> frameSeq = readImages(path);
      calibrator->addFrameSequence(frameSeq);
      frameSeqs.push_back(frameSeq);
      activeFrameSeqs.push_back(i);
    }
  }
  calibrator->checkerSize = ui->checkerSizeBox->value();
  calibrator->checkerCols = ui->checkerColsBox->value();
  calibrator->checkerRows = ui->checkerRowsBox->value();

  // Perform calibration
  calib = calibrator->calibrate();

  // Re-enable interface elements
  ui->calibrateButton->setEnabled(true);
  ui->listWidget->setEnabled(true);
  ui->saveButton->setEnabled(true);
}

void SLCalibrationDialog::on_listWidget_itemSelectionChanged() {
    std::cout << "on_listWidget_itemSelectionChanged" << std::endl;
  // If selection was cleared
  if (ui->listWidget->selectedItems().isEmpty())
  {
      std::cout << "empty" << std::endl;
      return;
  }

  // Stop live view
  killTimer(liveViewTimer);
  reviewMode = true;
  ui->snapButton->setText("Live View");

  int currentRow = ui->listWidget->currentRow();
  //ui->videoWidget->showFrameCV(frameSeqs[currentRow].back());
  std::string path;
  path = rootPath + "/" + ui->listWidget->currentItem()->text().toStdString();
  std::cout << "current path: " << path << std::endl;
  std::vector<std::string> names;
  cv::glob(path, names);
  if(names.size() > 0)
  {
      cv::Mat mat = cv::imread(names[0]);
      if(!mat.empty())
          ui->videoWidget->showFrameCV(mat);
  }
}`

thank you in advance.

jakobwilm, 2022-05-20:
Hi Mike.Haveyou verified your projector-camera linearity? This has come up a few times before. Linearity must be good before any hopes can be made. Sincerely

closed 2022-05-20 by jakobwilm

MikeDai2020, 2022-05-20:
thanks for your reply.