jakobwilm/slstudio #57: Frequncy enquirment

CHNJQB, 2023-02-06:
Hi ！
When I ran the software with my own data, I was unable to complete the final reconstruction.I suspect the frequency is wrong.
Does the software have requirements for the frequency of projected fringe?Is it necessary to project fringe patterns at specific frequencies?
Thank you very much.
[image]
[image]

jakobwilm, 2023-02-06:
Hello. Yes - of course.

closed 2023-02-06 by jakobwilm

CHNJQB, 2023-02-06:
> Hello. Yes - of course.

Could you please tell me the frequency requirement?
Thank you very much！

jakobwilm, 2023-02-06:
It needs to match decoder frequency. Studdy source code and theory.
This is not a software issue. Please stop using issue reports.

CHNJQB, 2023-02-07:
OK！Thank you for your sincere answer.