jakobwilm/slstudio #58: Guide on how to compile?

gamerpaddy, 2023-03-06:
Without being a lvl 9000+ Nerd on linux, how do you even compile and run it?
or on windows?

CWB1999, 2023-05-12:
I also want to ask how to compile and run the project in the Windows system. I opened the project with QT creator 4.11 and met an error like the following:
Error while building/deploying project SLStudio (kit: Desktop Qt 5.14.2 MSVC2017 64bit)
The kit Desktop Qt 5.14.2 MSVC2017 64bit has configuration issues which might be the root cause for this problem.

757dyar, 2023-06-27:
Dear Mr. Wilm,
Thank you very much for your work. Im working on an project which would really benifit from your codes.

I am using WSL2 (latest version) with ubuntu 22.04.  Mircosoft says that it can open open different linux apps such as chrome-linux, VLC-linux.

I followed the cmake procedures stated by Aaron in above comments.  My application was successfully made, but when open it it turns into a dark screen.

Do you have any idea on why that happens?  Thank you very much again!

Andy
[image]

closed 2023-06-27 by jakobwilm