jakobwilm/slstudio #65: calibration

nesquik011, 2025-04-01:
Hello , thank for sharing your work  i built the system on windows 10 it was not easy but everything works fine now even with Debug 

i implanted opencv camera , because i cant wait for the camera and projector to arrive , anyway i dont know what is wrong with the calibration in your video it draws markers on the chessboard , it does not do that :  

[image]

so kindly could you tell me what is wrong ? my chessboard is 16 x 10 , i picked in the settings to be 15 x 9 , and i also try 16 x 10 i thought maybe it could work but still it did not 

i know it wont work well with opencv but i am just testing functions and i will remove those modifications later

jakobwilm, 2025-04-02:
Hi.
It’s not clear to me what you are saying here: “I implanted opencv camera , because i cant wait for the camera and projector to arrive ,” - perhaps use an online translation tool.
Also given the little information I can only guess that perhaps your images are over or underexposed.

shadingi.png (view on web)<https://github.com/user-attachments/assets/9308d493-68fe-4d0b-aa4b-f3c0bc6407da>

shadingi.png (view on web)<https://github.com/user-attachments/assets/9308d493-68fe-4d0b-aa4b-f3c0bc6407da>

nesquik011, 2025-04-02:
> Hi.
> It’s not clear to me what you are saying here: “I implanted opencv camera , because i cant wait for the camera and projector to arrive ,” - perhaps use an online translation tool.
> Also given the [...]

I ordered a camera and projector, but they will take a long time to arrive. That's why I implemented an OpenCV function to open my webcam for testing. The problem is that the calibration does not work: when I use the chessboard, it doesn't detect it or even draw the markers. My chessboard is 16 x 10, but I set the settings to 15 x 9, and I also tried 16 x 10, thinking maybe that would work, but it still didn't. I know it won't work well with OpenCV, but I'm just testing functions, and I will remove those modifications later.

nesquik011, 2025-04-02:
@jakobwilm  do you understand what i said ?

nesquik011, 2025-04-02:
[image]  basically it does not show markers like this  @jakobwilm

nesquik011, 2025-04-03:
i thought maybe it does not pass the right rows and cols so i made CalibrationDialog.cpp prints the Rows and cols i set 

==========================================
Calibration parameters from GUI:
  Spacing: 15
  Rows: 15
  Cols: 9
  Pattern: "checkers"
==========================================
still did not work , so what is the problem ? @jakobwilm

jakobwilm, 2025-04-03:
Hi. That is not possible to say. Especially with your modifications. For calibration, data needs to be perfect and as shown in the video. You will need to do extensive debugging.

nesquik011, 2025-04-03:
@jakobwilm  lets say the shift phase patterns are not projected on the calibration board , and i snapped few images , is it going to stop opencv from detecting the chessboard ?  because i am thinking now maybe it needs to detect the shifting phase correctly to detect chessboard

jakobwilm, 2025-04-03:
Yes. Detection images are computed from the pattern images. Please study the code. I am going to close this.

nesquik011, 2025-04-03:
i think you did not get my question , lets say i snapped one set , if the pattern not visible , is opencv going to fail to detect the chessboard ?

closed 2025-04-03 by jakobwilm

Slashingbee, 2025-04-26:
can you provide binaries for windows, linux etc. tho i need just windows. Please provide the binaries, or at least if someone managed to compile them, send them to me somehow please.

nesquik011, 2025-04-26:
@Slashingbee  i know its not easy to build on windows , but i will help you as much as i cloud , my build is not pure there is some heavy modifications which i am too lazy to make things as original specially to run a webcam with opencv , so my build wont work with you , i did that so make sure things would work until i get better camera and learn the code more so give me your email or discord to chat

nesquik011, 2025-04-26:
@Slashingbee what camera are you going to use ? you would need to write code for it so the system could use it

Slashingbee, 2025-05-11:
@nesquik011 Im gonna use windows 11 (WIN10 support is fine) and ill be using a single pc webcam, if it can recognise windows cameras (with stuff like opencv) it would be fine, also i need the camera parameters to be changable or autodetected, and i want to be able to manually input the camera id or name. since its a generic camera. if you want to text me on discord, my discord is "Slash34" and u should get a reply.

Slashingbee, 2025-05-11:
also i know how to build it but i just dont think i want to spend the time for it, and also it'd be nice if someone got all the rep's for it as i dont know what modifications to apply to the repositories that is uses.