jakobwilm/slstudio #66: calibration sample

nesquik011, 2025-04-21:
i did study the code more , but i still have calibration problem , my physical chessboard is 16 x 10 , 15 mm 

what did i found out is in calibrationDialog.cpp it uses QSettings settings("SLStudio");   and in calibrationworker it uses QSettings settings;  so i thought maybe thats the problem so i changed it to QSettings settings("SLStudio");  as well and yet it did not work 

the settings i uses is 15 x 9 , spacing 15 mm , i hard coded the shading image and test it outside the system on python it worked fine 

here also the image 

[image]

and here the images of the snap , may you kindly test them in your environment see if it will work or no ?  i really cant find out what is wrong :

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

jakobwilm, 2025-04-21:
Hello.
This is not a code issue - please don't create issues on this.
Bright regions are saturated in your images so that will never work correctly.

shadingi_checkers_input.png (view on web)<https://github.com/user-attachments/assets/3dec2205-51d4-456f-aa31-0bbe10ee8400>

frame_00.jpg (view on web)<https://github.com/user-attachments/assets/42d4d672-ccd9-44d9-a221-c67fb53448b6>
frame_01.jpg (view on web)<https://github.com/user-attachments/assets/66e1bcf7-e3ba-4df0-8311-777511278243>
frame_02.jpg (view on web)<https://github.com/user-attachments/assets/83f7662d-6765-4f12-b557-eec3f0a55fd2>
frame_03.jpg (view on web)<https://github.com/user-attachments/assets/7336e16d-7edd-4ded-b388-9b335a0f4356>
frame_04.jpg (view on web)<https://github.com/user-attachments/assets/3e7be4dd-abaa-4d70-a617-ebe87f3e4878>
frame_05.jpg (view on web)<https://github.com/user-attachments/assets/37f975c0-06c8-4f58-a0b9-3fe1b308194b>
frame_06.jpg (view on web)<https://github.com/user-attachments/assets/5ca41799-0788-4e12-8670-bd88e031882e>
frame_07.jpg (view on web)<https://github.com/user-attachments/assets/85373e59-3d6c-46c8-ba94-f96c7b93736d>
frame_08.jpg (view on web)<https://github.com/user-attachments/assets/3beeaef0-0b0c-4947-9315-6c04316a4d5a>
frame_09.jpg (view on web)<https://github.com/user-attachments/assets/0cceba58-feb2-4c50-884f-7d2781068e30>
frame_10.jpg (view on web)<https://github.com/user-attachments/assets/4d542c1e-5e95-4234-9c1e-476e2b43f0c0>
frame_11.jpg (view on web)<https://github.com/user-attachments/assets/a7038bc8-f9d8-4b23-8012-2d5b95fa3696>

shadingi_checkers_input.png (view on web)<https://github.com/user-attachments/assets/3dec2205-51d4-456f-aa31-0bbe10ee8400>

frame_00.jpg (view on web)<https://github.com/user-attachments/assets/42d4d672-ccd9-44d9-a221-c67fb53448b6>
frame_01.jpg (view on web)<https://github.com/user-attachments/assets/66e1bcf7-e3ba-4df0-8311-777511278243>
frame_02.jpg (view on web)<https://github.com/user-attachments/assets/83f7662d-6765-4f12-b557-eec3f0a55fd2>
frame_03.jpg (view on web)<https://github.com/user-attachments/assets/7336e16d-7edd-4ded-b388-9b335a0f4356>
frame_04.jpg (view on web)<https://github.com/user-attachments/assets/3e7be4dd-abaa-4d70-a617-ebe87f3e4878>
frame_05.jpg (view on web)<https://github.com/user-attachments/assets/37f975c0-06c8-4f58-a0b9-3fe1b308194b>
frame_06.jpg (view on web)<https://github.com/user-attachments/assets/5ca41799-0788-4e12-8670-bd88e031882e>
frame_07.jpg (view on web)<https://github.com/user-attachments/assets/85373e59-3d6c-46c8-ba94-f96c7b93736d>
frame_08.jpg (view on web)<https://github.com/user-attachments/assets/3beeaef0-0b0c-4947-9315-6c04316a4d5a>
frame_09.jpg (view on web)<https://github.com/user-attachments/assets/0cceba58-feb2-4c50-884f-7d2781068e30>
frame_10.jpg (view on web)<https://github.com/user-attachments/assets/4d542c1e-5e95-4234-9c1e-476e2b43f0c0>
frame_11.jpg (view on web)<https://github.com/user-attachments/assets/a7038bc8-f9d8-4b23-8012-2d5b95fa3696>

nesquik011, 2025-04-21:
@jakobwilm  thanks for the respond , i did not know what to do thats why i opened a case , i apologize if i bothered you , but may you kindly keep it open to follow up ,  and maybe others who are stuck like me would find that useful as well

jakobwilm, 2025-08-22:
@nesquik011 I understand. But please familiarize yourself with the "Discussions" section on github. "Issues" is only for bugs.

closed 2025-08-22 by jakobwilm